lu7did/RDX-rp2040 #8: Callsign can not be updated

lu7did, 2023-04-26:
@Charudatt

When the code from github is downloaded and executed once it initializes the EEPROM information with the default data on
callsign, grid and other stuff.

If later this data is changed and the program recompiled the information isn't refreshed because a build version has not been
detected.

@Workaround: Change the configuration data, increase by one the BUILD number, compile and flash. Then execute. After that return the BUILD number to the previous, compile and flash. The data should be correct now.

lu7did, 2023-07-24:
Build #90 and up fixes this problem.

closed 2023-07-24 by lu7did

charudatt, 2023-07-24:
Could you also fix the time set issue , setting decimal values in time. We are +5.30 and the system cannot take this value

 

regards

charudatt – VU2UPX

 

Closed #8 <https://github.com/lu7did/RDX-rp2040/issues/8>  as completed.

lu7did, 2023-07-24:
Yes, Charudatt, it is the issue #22 which is still open.

charudatt, 2023-07-25:
some differences observed in v90 firmware with debug on-board

*	the Debug listing are not the same between the use of Si4732/CA2003
*	It does not show logging onto the WiFi and getting the time from NTP server as it used to show in the previous Listing
*	EEPROM does show the WiFi Credentials though
*	Setting the time zone to 0 does not show the UTC time on the display , as I feel it simply does not log onto the netwrok
*	Request - Please display the I2C status of the Si5351 also like shown for the Rx chip if used.

 

How do I change the default frequency from 70740 to 140740 ?

 

regards

charudatt

lu7did, 2023-07-25:
Hi Charudatt, it's the issue #22, still open.

lu7did, 2023-07-25:
Hi Charudatt,

In your note in red.

73 de Pedro, LU7DZ

> some differences observed in v90 firmware with debug on-board
>
> * the Debug listing are not the same between the use of Si4732/CA2003

The code is different, therefore the traces might differ. Remember the
DEBUG clause must be activated for the trace to occur, it's de-activated in
the repository code.

>
> * It does not show logging onto the WiFi and getting the time from NTP
> server as it used to show in the previous Listing

The standard source code at the repository is for the Raspberry Pico and
not the Raspberry Pico W, therefore all WiFi related functions are disabled
by default,
you might activate the support for the Raspberry Pico W uncommenting the
sentence

//#define RP2040_W             1  //Comment if running on a standard
Raspberry Pico (non Wireless)

>
> * EEPROM does show the WiFi Credentials though

Yes, because of the above, they are stored but not used.

>
> * Setting the time zone to 0 does not show the UTC time on the display ,
> as I feel it simply does not log onto the netwrok

Nope, because of the above NTP doesn't connect to the network, with the
Raspberry Pico standard the only time available is the manual
synchronization.

>
> * Request - Please display the I2C status of the Si5351 also like shown
> for the Rx chip if used.

Please log in as an enhancement for future implementation.

>
>
> How do I change the default frequency from 70740 to 140740 ?

You should change it while the firmware is running and it will be stored in
the EEPROM for the next power up, it will be reversed when you flash code
again though.
For a permanent change you should change the definition to the frequency of
your wish.

unsigned long freq = 7074000UL;

I'll post this note as a question in GitHub in case any other had a similar
problem.

charudatt, 2023-07-25:
Thank you for the reply

 

regards

charudatt – VU2UPX