lu7did/RDX-rp2040 #22: Time Zone Maths

charudatt, 2023-05-08:
Time Zone Math does not take decimals. eg. +5.3

lu7did, 2023-07-25:
This issue is still open.

lu7did, 2023-07-27:
Build 92 includes a fix for this issue.

closed 2023-07-27 by lu7did

charudatt, 2023-07-27:
It does not work, the time difference between UTC and IST (Indian Standard Time) is +5.30.

 

After making changes in the code I get a difference of 12 mins late

 

regards

charudatt vu2upx

charudatt, 2023-07-27:
It does not work

 

What the debug(Serial Monitor) window shows

 

[10:43:24] @setup Z=10:43 TZ=05:18 Local=16:01

 

What Changes I make in the code –

 

#define TIMEZONE   +5.50    //Buenos Aires, Argentina (GMT-3), change it accordingly using a fractional format VU-Land would be +5.50

 

Actual difference is +5.30 hrs.

 

regards

charudatt – VU2UPX

CT7ABA, 2023-07-27:
New Version shows local time at monitor instead of UTC

73 de CT7ABA