lu7did/RDX-rp2040 #24: After reboot the call sign disappears

CT7ABA, 2023-07-22:
After reboot the call sign disappears and at the screen i see only dots 

73 de CT7ABA
[image]

lu7did, 2023-07-22:
Hi Frederico,

This is a topic discussed on groups.io, but it worth to be documented here.

==============================[fix, long version]====================================
Proceed as follows:

Remove from your machine any source or binary file related to the RDX project, I meant, REMOVE all, don't try to save minutes at the expense of me lossing hours of debug chasing a ghost problem.
Execute the git clone https://github.com/lu7did/RDX-rp2040 command to recreate a new repository in your machine.
As per today you must be working with firmware VERSION 2.0 BUILD 87 (you can see it in the RDX-rp2040.h file). Do not touch it.
Using the Arduino IDE compile the project you just download, it must compile without any warning or error. Ensure your build environment reflects exactly what is stated in the section "Build environment" of the README.md documentation, including the previous proper installation of Earle Philtower's Arduino core libraries.
Once you build it successfully flash the Arduino Pi Pico board with the firmware.
Upon execution it will show up the default callsign and grid (LU2EIC/GF05).
Using the IDE go to the RDX-rp2040_User_Setup.h file and change the following statements only 
#define MY_CALLSIGN "LU2EIC" replacing LU2EIC by your callsign, leave no blanks and ensure you properly open and close the quotes.
#define MY_GRID "GF05" replacing GF05 by your grid using a 4-letter grid format, same precautions than with the callsign.
Compile and flash the code. Upon start the display will still show LU2EIC/GF05 because it's stored in EEPROM.
Power off the board.
Press the TX switch. While keeping it pressed power on the board. The LEDs will play a funny dance while you keep the TX button pressed.
Release the TX button. The board should now came with your callsign and grid. 
If fixed nothing else needs to be done.

If the above doesn't work as intended be sure the "#define DEBUG 1" line in the RDX-rp2040_User_Setup.h file is uncommented, if already uncommented
repeat the above procedure with the Arduino IDE Serial Monitor (found at Tools/Serial Monitor) open, capture in a file whatever is set there. 

If the DEBUG line was commented out and therefore nothing is shown in the Serial Monitor then uncomment it, compile, flash and repeat the above procedure starting with the "Upon execution it will show up...." line.

lu7did, 2023-07-24:
The callsign appears on the next reboot, but build #90 closes the issue by correctly updating the callsign/grid after a reboot and EEPROM reset.

EEPROM can be reset by turning the power on while keep pressed the TX button.

73 de Pedro, LU7DZ

closed 2023-07-24 by lu7did