lu7did/RDX-rp2040 #25: TFT Monitor Problem

CT7ABA, 2023-07-22:
TFT Monitor Problem, after changing line 21 (#define ROTATION_SETUP 3) at the TFTgui.h the rotation is ok but the touch points are not anymore calibrate.
Is there a possibility to recalibrate? and how can it be done?

73 de CT7ABA

https://github.com/lu7did/RDX-rp2040/assets/26884406/4df274bd-ff3c-4fb9-979c-10c0a8cec5d4

lu7did, 2023-07-22:
Hi,

A programatic way to re-callibrate the display is still in the pending queue. 

Not sure why you change the rotation define, not sure what the effect of that is either, never tested that scenario.

You can experiment by modifying the default calibration data, which is located in TFTgui.cpp 

uint16_t calData[5] = { 162, 3712, 292, 3458, 7 };

If you look at the library TFT_eSPI-Master which you should have installed already (as part of the Arduino environment) you'll find some examples dealing with the calibration of the TFT, you might want to play a little with them. It won't be a bad idea as a mitigation till a proper function is developed to have some auxiliar utility which provides the "right" calibration data for a given display. If you develop such a thing it will be a very useful collaboration I will gladly accept.

73 de Pedro, LU7DZ

CT7ABA, 2023-07-25:
Hi,
Coming back to your question due to why the rotation of the TFT monitor.
(Not sure why you change the rotation define, not sure what the effect of
that is either, never tested that scenario.)
The TFT monitors I bought from China are delivered rotated at 180°
Due to a calibration program, sri I'm not a Ham with such good skills in
programming C like you, I'm more a usufructuary of such good programmer's
as you are.

73 de Frederico CT7ABA

Juli 2023, 23:33:

CT7ABA, 2023-07-26:
It looks like that in the examples folder of TFT_eSPI there is already a calibration script that works on RP2040. I did run it and it looks pretty good. Pedro maybe you can use it to include it on your project.
C:\temp\Arduino-Libraries\TFT_eSPI\examples\Generic\Touch_calibrate\Touch_calibrate.ino

73 de Frederico CT7ABA

lu7did, 2023-07-27:
Yes, have it, didn't had time to integrate into the code, it's on the roadmap/queue.