lu7did/RDX-rp2040 #26: unknown Jumper funktion

CT7ABA, 2023-07-25:
Hello Charudatt, can you help me understand the funktion of J8 and J10 on your PCB
Thanks for your Help

73 de Frederico CT7ABA
[image]

lu7did, 2023-07-25:
I believe it's a board mistake, you should left it open. There is another jumper (J10) also labeled as RUN which bypass the Q2 amplifier which should be left open too.

73 de Pedro, LU7DZ

CT7ABA, 2023-07-25:
TNX Pedro for your informacion.
I'm facing a lot of problems and I'm wont to be sure that I didn't forget or made something wrong during the asambling of the board.

73 de Frederico CT7ABA

lu7did, 2023-07-25:
Well, this is part of the fun, isn't it?

There are two schools of building crafts. 

One places component by type and in number order. So first all capacitors, then all resistors, followed by electrolitics, inductors, transistors, IC and finalizing the miscellanea. It has the advantage of allowing a "bulk" building (you place all resistors and solder all of them on a single pass). You need a support to build this way because the board itself become uneven and components might be damaged if pressed on the table.

The other follows a "functional" approach. You build first the power sub-system, then test it. And you continue block by block testing each one as finished. It make unlikely that you made a mistake to be discovered at the very last that, eventually, might ruin some component because of it (i.e. running excessive voltage on a low voltage rail).

I do prefer the second, but it's just a matter of selection.

In both cases I do check, component by component with a component testers (really inexpensive for the huge boost in building capabilities) the component is ok and it's value is within nominal ranges. Frankly tired to spot a resistor very out of value, a capacitor open or an electrolitic with way too much loses. Not to mention transistors shorted or opened. It save you tons of time debugging and certainly the royal PITA of removing a well soldered component to replace it.

But either method before to power it up I carefully follow the circuit with a magnifier lens and cross with a resalter pen on a clean copy of the circuit component by component, and check whenever possible with a tester good connections between different parts of the board.

Personally I enjoy the journey in full, it's a land of no time of no worries while it last.

73 de Pedro, LU7DZ

closed 2023-07-25 by lu7did

CT7ABA, 2023-07-25:
Thanks for the lesson, will try to follow your preferred version next time ;-)

73 de Frederico CT7ABA