lu7did/RDX-rp2040 #29: No WIFI Conection

CT7ABA, 2023-07-25:
RDX does not connect to Wireles Network
#define RP2040_W    is uncomented in the RDX-rp2040_User_Setup.h file but still no connection, DEBUG shows the following information:
[00:00:33] @checkAP Failed to connect to ssid(Gaiaz1)
[00:00:33] @getClock NTP server pool ntp1(132.163.97.1) ntp2(pool.ntp.org)
[00:00:53] @getClock Attempt to connect NTS server failed rc(True)
[00:00:53] @resetAP WiFi desconnection in progress from IP(0.0.0.0)

After unplug the RP2040 from Charudatt's PCB and start the RP2040 without plug them in to the PCB it connects normally to the wireless network. The DEBUG shows that connection was successful.
[00:00:04] @tft_syncNTP WiFi NTP time sync ntp(132.163.97.1,pool.ntp.org)
[00:00:10] @checkAP Connected to ssid(Gaiaz1) IP(192.168.1.102) rssi(-58)
[00:00:10] @getClock NTP server pool ntp1(132.163.97.1) ntp2(pool.ntp.org)
[18:46:32] @getClock NTP time synchronization completed now is Tue Jul 25 18:46:32 2023
[18:46:32] @resetAP WiFi desconnection in progress from IP(192.168.1.102)
[18:46:33] @SI473x_setFrequency Frequency set to 7.074. KHz

73 de Frederico CT7ABA

CT7ABA, 2023-07-25:
It looks like that there is a conflict with the SI5351 then after unplug the SI5351 TRX connect's witout problems to the Network.

lu7did, 2023-07-25:
Weird...... never experienced such a problem.