lu7did/RDX-rp2040 #31: Strange signals/output at TFT-Monitor

CT7ABA, 2023-07-27:
Strange output at TFT-Monitor, I cant see any FT8 signals and no decode, something similar to your foto at:
https://github.com/lu7did/RDX-rp2040#rdx-board-from-charudatt-vu2upx---si473x-chipset-support

73 de Frederico CT7ABA
[image]

lu7did, 2023-07-27:
Please see the fix in the hardware changing GPIO2 with GPIO3

CT7ABA, 2023-07-28:
Still the same with Built 95.
1.) To clarify if the problem comes from a defect SI473* (I do not have the possibility to check it with an Oscilloscope and signal generator) I assabled one more RDX but the problem exists also on the new one.
2.) Output Power is roundabout 2W at the moment, have to optimize the LPF
3.) people are seeing me with the 2W in Europe and USA, TX works pretty good. The signal is much better then with the ADX-UNO.

The only problem is no RX and strange signals on the Monitor.

73 de Frederico CT7ABA

[image]
[image]
[image]

lu7did, 2023-07-30:
Yes, I'm able to reproduce the problem. It's nothing related to the decoding algorithm, loading the same firmware on a ADX based board (CD2003GP receiver) works just fine. Weirdo signals at the spectrum can be a callibration on the map between signal strenght and color, but a strong (strong as a station in the same room) 7 MHz signal is decoded erratically and not shown in the display. Looking with an oscilloscope the input of the Si473x (pin 8) has a 20-30 mV pk-pk 7 MHz signal, still at the output at pin 16 erratically shows erratically too a strong "hum-like" signal with a 7 MHz RF signal on top of it in the order of 200 mV pk-pk. At times it shows a more clear 1 KHz signal (with slight straces of a 7 MHz signal on top). The amplifier (Q2) has also a strange behaviour, when erratically works the output is a "chopped" mostly clean 1 KHz signal with the negative lobe saturated, probably this will be cured removing C8 with a short and placing a 10K resistance between the C13 capacitor and the Q2 base to reduce the gain. But the problem is all of this being erratic. Working actively on this issue. Leads I'm following is to check the proper wiring of the Si473X in the board, some noise because of a ground loop or auto-oscillation somewhere and the reset of the Si473x leading for no further or erratic output.
This is the main focus of my limited time at this poing.

73 de Pedro, LU7DZ

lu7did, 2023-07-30:
[image]

Yes, I'm able to reproduce the problem. It's nothing related to the decoding algorithm, loading the same firmware on a ADX based board (CD2003GP receiver) works just fine. Weirdo signals at the spectrum can be a callibration on the map between signal strenght and color, but a strong (strong as a station in the same room) 7 MHz signal is decoded erratically and not shown in the display. Looking with an oscilloscope the input of the Si473x (pin 8) has a 20-30 mV pk-pk 7 MHz signal, still at the output at pin 16 erratically shows erratically too a strong "hum-like" signal with a 7 MHz RF signal on top of it in the order of 200 mV pk-pk. At times it shows a more clear 1 KHz signal (with slight straces of a 7 MHz signal on top). The amplifier (Q2) has also a strange behaviour, when erratically works the output is a "chopped" mostly clean 1 KHz signal with the negative lobe saturated, probably this will be cured removing C8 with a short and placing a 10K resistance between the C13 capacitor and the Q2 base to reduce the gain. But the problem is all of this being erratic. Working actively on this issue. Leads I'm following is to check the proper wiring of the Si473X in the board, some noise because of a ground loop or auto-oscillation somewhere and the reset of the Si473x leading for no further or erratic output.
The attached image shows my results (more intense spectrum colors is because I'm using a more sensible color map than the one published in order to detect fainter signals).
This is the main focus of my limited time at this point.

73 de Pedro, LU7DZ

CT7ABA, 2023-07-30:
Hello Dr. Pedro.
yes that's identical copy of my monitor.
I have done also some smal tests and this are my results:

1.) unsolderd SI473x and connected a GD2003 flying Dutch circuit.
Result= After connecting the cables of the flying dutch and without
powering it on i got the same strange output on the monitor.

2:) After powering on the Flying Dutch.
Result= Start to receive more noise on the Monitor but still no decoded
signals.

2:) Sending FT8 CQ from my Transceiver TS-480.
RDX received my signal and display it at the monitor but with very poor
signal (-24) and it does not decode it every period.

73 de Frederico CT7ABA

Juli 2023, 15:14:

lu7did, 2023-07-30:
Hi,

Nor sure what the CD2003GP circuit in "flying dutch" (holandés volador hihi..) means, I've had considerable testing with the circuit using the CD2003GP both using a daughterboard and an ADX-UNO and a wired version of the RDX circuit https://github.com/lu7did/RDX-rp2040/raw/main/docs/PDX_V1.0_Schematic.jpg and it does receive signals very well, even moderated to faint ones (3 to 6 dB less sensible with my FT2000 transceiver), so this configuration works Ok for sure, extensible testing done.

To this date I had the information that the board has been tested successfully, but it seems not to be the case, at least not with the depth I thought. No permanent damage done.

I've posted a query on the Si473x library (PU2CLR) for possible issues on the library with this end result, I'm suspecting on some random reset of the Si473x, perhaps something wrong on the firmware that handles it or some circuit ground loop or noise with improper filtering. 

Eventually I'll find out, this circuit isn't much different than others that works Ok, so the bug might be hidden or obscure, but eventually will emerge.

73 de Pedro, LU7DZ

lu7did, 2023-07-31:
I think I found the problem, I'm working on what the best solution might be. It's a board issue,  a couple of them actually.

With the board as it is:

Si473x RF Input (pin 8) with a nearby signal of 7 MHz.
[image]

Colector of Q2
[image]

This is pure random noise with some 7 MHz RF on it.

The Si473x GP02/INTB line is open, high impedance, probably with a random noise from the RF environment, with a brief short circuit between this line and GP03/DCLK the noise disapears....

[image]

When a signal is created at 7.074 MHz the colector of Q2 is now
[image]

Which is a distorted audio signal, but much cleaner. Placing a 22K resistor between C13 and the base of Q2 (reducing the gain of the stage) the audio doesn't exceeds the polarization point and the signal isn't distorted.

[image]

Now the (local) decoding is solid
[00:39:15] @ft8_run -RX w[1/60248]-[7.074 KHz]-(0/0) (V=255)-
[00:39:28] @ft8_run [00] w[1] 1731  -13 CQ LU7DZ GF05
[00:39:30] @ft8_run -RX w[0/60160]-[7.074 KHz]-(0/0) (V=255)-
[00:39:45] @ft8_run -RX w[1/60160]-[7.074 KHz]-(0/0) (V=255)-
[00:39:58] @ft8_run [00] w[1] 1931  -11 CQ LU7DZ GF05
[00:40:00] @ft8_run -RX w[0/60160]-[7.074 KHz]-(0/0) (V=255)-
[00:40:15] @ft8_run -RX w[1/60160]-[7.074 KHz]-(0/0) (V=255)-
[00:40:28] @ft8_run [00] w[1] 1931   -8 CQ LU7DZ GF05

The antenna used for this test is a 10 cm piece of wire, so that is the typical SNR I got with this configuration.

Have still to figure out how to find a board fix/firmware fix combination to reproduce the manual "surge" that solves the problem, and work with the calibration of the waterfall to the available levels of the Si473X (plus some work with the volume which is at the maximum at this point.

Will keep working on the issue, but there is light at the end of the tunnel (and hope it's not a train :-) ),

73 de Pedro, LU7DZ

CT7ABA, 2023-07-31:
TNX for your detailed information, I think the light is not a train. The flying dutchman is a Ghost ship legend and my circuit on the breadboard looks lik it :-)).
I'm very anxious to see your baby running GL.

By the way, are they somewhere Gerber Files for your PDX?
TNX for the unpaid work you are doing and the time you ar spending for the Hamradio comunidad

73 de Frederico CT7ABA

lu7did, 2023-08-12:
I'm discussing on a separate forum with the Ricardo (PU2CLR) who is the creator of the Si473X library and a well known expert on that chip used for amateur purposes. The latest evolution on the issue, just in case anybody can replicate this behaviour, is as follows:

- Without the Si5351 BoB everything works ok, including the Si473x. This points in the overall direction of some I2C bus issue.
- With a 4K7 pull-up resistor on the I2C bus the Si5351 still works, the Si473X doesn't exhibit the noise output but doesn't yield any output either.

Next test is to use a 10K instead, Ricardo says the I2C bus of a Si5351 is 22K and the Si473X needs anywhere between 3K3 and 10K, thus the 4K7 pullup places the paralell impedance close to the limit. I do need to test with a 10K pullup in both SDA and SCL of the I2C bus and see.

Another test to make is if this behaviour happens with an external +12V source, at this point I'm providing +5Vcc from the USB port of my Mac (shared with the serial port) and the rest is feed by the 3.3V provided by a local regulator.

CT7ABA, 2023-08-12:
Hallo Pedro, do you remember my problems with some of my SI5351? (Wifi did
not work with some of my SI5351)
The problem is derived to an address conflict, maybe you found a solution
for both problems.
I received yesterday a bunch of new SI473X and tested to change the i2c
from 0x11 to 0x63 address by connect SEN (Pin 10) to 3,3V but it didn't
help.
Hope you have found a solution so that your baby can move from beta status.

I love this little special rig.

73 de Frederico CT7ABA/DM3FG

Aug. 2023, 17:40:

CT7ABA, 2023-08-12:
Hello pedro,
I have seen that CLK1 of the SI5351 is always on (7,074) is it possible to
set it to off? This could be also a problem, mybe there is some RF from the
SI5351 entering into the SI473X AMI pin8

73 de Frederico CT7ABA/DM3FG

Aug. 2023, 17:40:

lu7did, 2023-08-12:
Some breakthru on the issue.

The following mods seems to have fixed the problem:

- 10K pull up resistor from Si5351 pin 3 (SDA) to +3.3V.
- 10K pull up resistor from Si5351 pin 4 (SCL) to +3.3V.
- Add a 22 pF capacitor in paralell with C11 (22pF).

This mods makes the receiver to initialize properly and operate correctly, there are some odd displays in the waterfall (as if the bandwidth of the Si473x were reduced) but I'm working on it. The main issue has been solved.

lu7did, 2023-08-12:
This issue has been fixed, remaining issues are tracked on different issues.

closed 2023-08-12 by lu7did