lu7did/RDX-rp2040 #34: RDX-rp2040 board wiring fix needed

lu7did, 2023-08-12:
The following fixes are needed on the RDX-rp2040 (Si473X chipset based) board.

- Pull-up resistor from pin 3 of Si5351 (SDA) to +3.3V.
- Pull-up resistor from pin 4 of Si5351 (SCL) to +3.3V.
- Additional capacitor 22 pF in paralell with C11 might (optionally) be needed.

CT7ABA, 2023-08-12:
Hi,
1.) All of my SI5351 modules have already a 10K resistor and the problem still remains. 
2.) The red one starts OK with the RDX but no reception.
3.) The purple ones don't let the RDX obtain an IP-Adress from the network  (RP2040 does not connect to the WITH Network)
4.) The purple one with the White point is one of the modules I used with Barbs uSDX, the ones we had to rework by unsoldering some components ( https://antrak.org.tr/blog/usdx-a-compact-sota-ssb-sdr-transceiver-with-arduino/)
This one works fine with t RDX as you can see at the picture.
73 de Frederico CT7ABA/DM3FG 

[image]
[image]
[image]
[image]

CT7ABA, 2023-08-13:
To be sure that the reworked SI5351 works,  I'd reworked one more module and the result is it works fine it connects to the network and the reception is also OK.
73 de CT7ABA/DM3FG 

[image]

lu7did, 2023-08-13:
Hi, 

Did you place the 22pF capacitor in parallel with C11?
Did you upgrade to build 98?

Remember the waterfall isn't working properly at this point, check the reception though, it should work.

Regards, Pedro.

CT7ABA, 2023-08-13:
Hi,
No I didn't, I did only change the SI5351 module and it works fine with the
build 95.

73 de Frederico CT7ABA/DM3FG

Aug. 2023, 03:19:

CT7ABA, 2023-08-13:
Hi,
After installing build 98 and inserting the adicional C to C11 no more
reception and no network connection.

Aug. 2023, 03:19:

CT7ABA, 2023-08-13:
Hi.
Strange things happens.
After changing the SI5351 and the installation o the 22pF capacitor I do
not receive anymore also when the SI5351 is unplugged.
(Settings in RDX-rp2040 for the SI473X are done)

I will send you tomorrow a brand new SI5351 module for you to see the
modules I use.

With the build 95 I do not have any problems with my reworked modules.

73 de Frederico CT7ABA/DM3FG

Aug. 2023, 03:19:

lu7did, 2023-08-13:
Remember the standard upload status is:

* CD2003GP receiver.
* No debug.
* Raspberry Pi Pico std 

In order to change it to Si473X you need to uncomment the line **#define RX_SI473X       1** in RDX-rp2040.h
In order to change the DEBUG you need to uncomment the line **#define DEBUG               1** in RDX-rp2040_User_Setup.h
In order to change to Rpico Wireless you need to uncomment    **#define RP2040_W         1** in RDX-rp2040_User_Setup.h

It is not probably a good (or at least easy solution) but the stable version is still the CD2003GP using a non wireless pico based one.

73 de Pedro, LU7DZ

CT7ABA, 2023-08-13:
Yes, that's all done but I still facing problems between SI473X, SI5351 and
Wireless.
Yes I think also the best is to use the version with CD2003GP instead of
the SI473X ore maybe to use a I2C extender to avoid those conflicts.

I do have only those boards where you can use only the SI473X chips, I have
to ask Charudatt if they won't to sell mi some.

73 de Frederico CT7ABA/DM3FG

Aug. 2023, 17:08:

lu7did, 2023-08-13:
Don't throw the towel yet.

Enjoy the ride, it's part of the fun.

Charudatt´s board works as long as all the fixes mentioned in the README.md are done (two groups of them reflected in two issues here).

I would like to be sure your boards works with the Si4732 chip, this is the one I have and it works, I'm having it soldered on the board, not as a BoB (I don't know nor I can debug if there is any issue with a BoB connection in that case).

I2C has no issues as long as you place the 10K pull-up as indicated, I've my board powered from the USB port of my PC and it works, the Mac is legendary for having very picky behaviour with the USB ports in terms of load, and it works.

You should just buy the chips and solder them directly on the board.

Once it works I would help you with the fun of debugging another board with the CD2003GP on it :-)

73 de Pedro, LU7DZ

CT7ABA, 2023-08-13:
Hi,
I do not throw the towel away, I thought that you are using a different PCB
then I do.
The mods are all done and with version 95 I can receive and transmit but
only if I use the SI5351 modified as I showed you.

In the moment I'm trying to compile build 98 and 99 but I'm running into
this error:

c:/users/ich/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe:
C:\Users\Ich\AppData\Local\Temp\arduino\sketches\5A5DBF09FFFF247D932EDC6F7E8B094B\sketch\Si4735.cpp.o:
in function `_Z9getSignalv':
Z:\Amateurfunk\LU7DID\RDX-rp2040-main_build98\RDX-rp2040-main\src\RDX-rp2040/Si4735.cpp:278:
undefined reference to `rssi'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

PS: build 95 compiles without problems

73 de Frederico

Aug. 2023, 19:06:

CT7ABA, 2023-08-13:
Hi,
The compiling failure I reported in my last reply, happens only if I
uncomment the Debug line on RDX-rp2040_User_Setup.h

73 de Frederico CT7ABA

Aug. 2023, 19:06:

lu7did, 2023-08-13:
Sorry, all errors are silly, some are more difficult to fix. 

This was easy, build 100 has it corrected.

73 de Pedro, LU7DZ

lu7did, 2023-08-13:
Incidentally, I've three development boards. Two based on the CD2003GP and another with Si473X, the later being the last to develop (or in other words the one most behind) therefore is the one I'm placing more focus on. The boards based on the CD2003GP are really modified ADX boards and thus less prone to have hardware problems, so most of the problems are on the firmware. On the Si473X board I have plenty of problems in both fronts, hardware and firmware, and sometimes interacting in very strange ways (appearing to be a firmware bug and being a hardware issue or vice versa).

In any case you're helping a lot on the debugging of the Si473x board, so thankful for your aid.

73 de Pedro, LU7DZ

CT7ABA, 2023-08-13:
I have to thank you for this wonderful rig your are developing.

By the way a good news: the reception is working now and also the Wireles
with build 99 but with the si5351 board I modified without the two Mosfet's
and the 10K resistors.

73 de Frederico CT7ABA

Aug. 2023, 00:24:

lu7did, 2023-08-14:
This information is priceless, most of the time I'm chasing whether a problem lies on the firmware or hardware side, this splits the problem into a neat hardware problem (as it works correctly in at least one HW configuration). The modifications might be related either noise or consumption. More data would make the diagnostic converge into some more robust as a clue.

73 de Pedro, LU7DZ