lu7did/RDX-rp2040 #35: Poor RX sensibility

CT7ABA, 2023-08-14:
Hi,
RX sensibility has dropped down and RX Report shows allways -24

73 de Frederico CT7ABA

lu7did, 2023-08-15:
The reported sensibility, actually the decoded SNR, isn't connected to the GUI representation yet. the -24 is the minimum of the decoding algorithm (sort of a representative of 0) and used as a placeholder.

closed 2023-08-15 by lu7did

CT7ABA, 2023-08-15:
Hi.

OK, but it does not explain why I do not receive more Stations, in some
periods I recive some Stations and in some I do receive not only one but
there are hundreds of stations calling (I could saw it on my ts480)

TX is working fine, they see me in Europe and in the states also.

Aug. 2023, 14:13:

CT7ABA, 2023-08-15:
Hi.

OK, but it does not explain why I do not receive more Stations, in some
periods I recive some Stations and in some I do receive not only one but
there are hundreds of stations calling (I could saw it on my ts480)

TX is working fine, they see me in Europe and in the states also.

73 de Frederico CT7ABA
[image]

CT7ABA, 2023-08-16:
Hi.

You closed the issue #35 but they are still problems related to the
reception.

The TX part of Your RDX is working great but the RX does not. I'm even not
able to make a qso between the RDX and my comercial rig (CT7ABA / DM3FG)

1.) Calling CQ from the RDX, is been seen on my comercial rig.

2.) Calling CQ from my Comercial Rig, is not being seen on the RDX.

3.) Calling CQ on RDX, other stations are responding I can see it on my
Comercial Rig, but not on the RDX.

Could it be a frequency discrepancy between the RX part and the TX part of
the RDX? I don't know if it is so because sometimes (rarely) I see the same
station on both rig's.

The original SI5351 module is sent out today because we had a Bank Holiday.

PS: I did calibrate the RDX before I tried it out.

73 de Frederico CT7ABA

Aug. 2023, 14:13:

lu7did, 2023-08-16:
The issue was stated because the board RX didn't worked, the Si473X entered a high Z mode where the demod signal were not present at the output, it was because of the Si5351 interfering with the Si473X over the I2C bus and was solved with the additional pullup. Also a slow starting OSC on the Si473X were pulled with the additional capacitor. When performed that fix the RX start to operate and the test case for RX worked correctly.

The test case I use to test the firmware is to TX with a small short wire (50 cm or so) and receive it on my station and capture my station regular transmission to be reflected. Both tests works, showing the firmware encodes and decodes correctly. I didn't run any lab test on the RX part of the board so far to see the actual sensibility, it won't hurt to bring it to some lab and get results.

I meassure (reflected in the S-meter) the RSSI is really high with a local signal.

The debug of the Si473X board is a multifront effort, I'm concentrating at this point on the firmware of it and debugging the hardware if and only if doesn't allow the firmware to operate correctly (like in missing lines, biases, lack of pull-ups, decoupling, oscilator, etc). 

However, I saw a photo of your screen with multiple decoding on it, how it was obtained?

73 de Pedro, LU7DZ

CT7ABA, 2023-08-17:
Hi  tnx for your info.
I'm not anymore shure, if the problem is a poor reception or a frequency
deviation between SI473X and si5351, calling at one frequency and receiving
far-away from tx frequency

The RDX is connected to a Endfeed with roundabout 22 meters.

73 de Frederico CT7ABA

Aug. 2023, 22:34:

lu7did, 2023-08-17:
I'm not really sure, it might be something specific to your configuration. I don't know if you have instruments (i.e. a scope) to see the frequencies involved in the Si5351.

If you're receiving signals and decoding them, despite if it's getting your own, it is not a Si473x issue.

You should be able to hear, very distinctively your own emission on your nearby receiver, and also to see the spectrum of what you're emitting (which should be quite strong) on the WSJT-X associated with that receiver. You'll have an idea on whether your Si5351 is shifted or not (it emits at this point a 1500 Hz + fLO). The other way around with the Si473X.

I'm doing all my tests with just a 50 cm wire so a 22 mts endfeed is an overkill for that kind of local test.

73 de Pedro, LU7DZ