lu7did/RDX-rp2040 #36: WEBTOOLS not working

CT7ABA, 2023-08-14:
Webtools / Webserver does no work.

#define WEBTOOLS is uncommented in the rdx_rp2040_user_setup.h and UP/DOWN buttons are pressed during restart but nothing happens. 
I can see on the monitor that RDX connects to WiFi Network, it obtains an IP address it sets the time, during this period the RDX is responding to a continuous ping from my notebook but after time setting it disconnects from WiFi network.

73 de Frederico CT7ABA

lu7did, 2023-08-15:
Webtool can not be used while the transceiver is operating, so when activated the transceiver operation is stopped and has to be finished by reset of the transceiver.

When enabled there is an icon for it at the top bar (a terminal icon) which gets activated, by pressing it the internal Web can be accessed as per the manual.

Not sure it's a practical feature, it can be done though, so it's there.

73 de Pedro, LU7DZ

CT7ABA, 2023-08-15:
Hi
It does not work or I'm doing mistakes,  this is the way I doe's understand
and the way I try to do it:

1.) Uncomment line 23 on rdx_rp2040_user_setup.h

2.) Reboot the RDX and pressing up & down knobs

3.) Try to activate the icon (not possible)

4.) Try to access the internal Webserver with the Webbrowser by typing
ip:port (192.168.1.102:8000) but nothing happens.

73 de Frederico CT7ABA

In my case it's a very practical tool due to my two callsigns.

Aug. 2023, 14:18:

lu7did, 2023-08-16:
I guess that between step 1 and 2 of your procedure you recompile and reflash the firmware, right?

CT7ABA, 2023-08-16:
yes, right
First the modification then compiled and upload.

CT7ABA, 2023-08-17:
> The way it works is.
>
> When enabled at firmware build time (which apparently you did). Main
> difference showing that is that the Web icon instead of being crossed will
> be opaque.
>
> Once the setup has been completed when clicking on the web icon (which
> will be recognized at the end of the RX window) it will stop being opaque
> and the text window will say "Web configuration tool active". At that
> momment the board will connect with the WiFi router (the same configured
> for the NTP sync) and will listen for requests for the Web port.
>
> Once finished on the web side the board transceiver function will reset,
> if not you should reset the board as any relevant change should be in
> EEPROM by then.
>
> 73

Hi
It is not possible to activate it, the only icon's I could activate are
DATALOGGERUSB and the one with i (the third from lsft) but it does not
export ore activate a drive on my Windows PC.

73 de Frederico  CT7ABA

Aug. 2023, 16:12:

lu7did, 2023-08-30:
Please run a DEBUG session trying to capture the setup phase (this is accomplished by starting the Serial Monitor and then repower the board).