lu7did/RDX-rp2040 #37: DATALOGGERUSB

CT7ABA, 2023-08-30:
Hi.
Problem exporting logfile.
On RDX-rp2040_User_Setup.h line 28 is activated, firmware recompiled and uploaded to the RDX but still not possible to obtain the logfile.

1. USB icon accepts the touch
2. The blue icon with the exclamation sign changes color.
3. At the reception window appears "USB export available"

And thats it.

73 de Frederico, CT7ABA 

[image]
[image]
[image]

lu7did, 2023-08-30:
Hi Frederico,

This is another half-cooked "feature" of RDX, I did implement it, tested
with few use cases and left until further use.

I assume you placed an USB stick or memory, have a firmware compiled with
all the signals needed to activate and after bootup you activate the
feature.

¿Does your machine show the USB drive when doing that? Nothing will happen
on the RDX, there will be (should be) a file on that drive in your PC when
doing so. ¿Did you had contacts to show? (This feature doesn't have
persistence between power cycles of the RDX board, not yet at least)

If you comply with all the above please run a session with DEBUG activated
and send me the result in order to fix the problem.

73 de Pedro, LU7DZ

lu7did, 2023-08-30:
I believe this aspect isn't particularly well documented but ... ¿did you leave space for a File System (FS) on the flash memory?
[image]

On Tools/Flash Size at least 64K of the flash should be allocated to a file system (would be read as

CT7ABA, 2023-08-30:
Hi,
Yes it is compiled with the option to leave 64K for flash but it does not
mount a drive on my PC, could it be that it recognises that there is no
data to be written to the file?
I will test it again later after doing a qso.

73 de Frederico, CT7ABA

Aug. 2023, 13:14:

lu7did, 2023-08-30:
Please initialize the Serial monitor and then reboot the board so the messages during the setup phase are shown, these messages would tell me whether the feature are actually activated or if there is some problem with it.

Thanks, Pedro LU7DZ

CT7ABA, 2023-09-01:
Hi,
here the requested Debug output.

RDX Raspberry Pico Digital Transceiver
Version(2.0) Build(102) (c)Pedro E. Colla (LU7DZ)
[00:00:01] @setup Overclock activated
[00:00:01] @setup Multicore activated
[00:00:01] @setup rp2040 W processor support activated
[00:00:01] @setup RX Si473x based receiver defined
[00:00:01] @setup Support for TFT IL9488 activated
[00:00:01] @setup ADIF loogbook support activated
[00:00:01] @setup USB ADIF export activated
[00:00:01] @setup Initial band_slot=4
[00:00:01] @setup Firmware build (102)
[00:00:01] @getTZ timezone=-0.00 tzh=0 tzm=0
[00:00:01] @INIT I/O setup completed
[00:00:01] @INIT EEPROM reading completed EEPROM build(102) Firmware build(102)
[00:00:01] @readEEPROM Start Band_slot(4)
[00:00:01] @readEEPROM  completed Band_slot(1)
[00:00:01] @Slot2Freq Slot=1 band=40 index=2 freq=7074000l
[00:00:01] @SI473x_setFrequency Frequency set to 7.074. KHz
[00:00:01] @INIT Band_slot(1) freq(7074000l)
[00:00:01] @initSi5351 initialization rc(1)
[00:00:01] @setup initial key configuration up(True) down(True) tx(True)
[00:00:02] @Slot2Freq Slot=1 band=40 index=2 freq=7074000l
[00:00:02] @SI473x_setFrequency Frequency set to 7.074. KHz
[00:00:03] @tft_setup Initialization completed
[00:00:03] @setup TFT sub-system ready
[00:00:03] @setup TFT default calibration adopted
[00:00:03] @setup Initializing ADIF logbook sub-system Log(True)
[00:00:03] @tft_syncNTP WiFi NTP time sync ntp(132.163.97.1,pool.ntp.org)
[00:00:10] @checkAP Connected to ssid(Gaiaz1) IP(192.168.1.102) rssi(-74)
[00:00:10] @getClock NTP server pool ntp1(132.163.97.1) ntp2(pool.ntp.org)
[12:37:19] @getClock NTP time synchronization completed now is Fri Sep  1 12:37:19 2023
[12:37:19] @resetAP WiFi desconnection in progress from IP(192.168.1.102)
[12:37:20] @Slot2Freq Slot=1 band=40 index=2 freq=7074000l
[12:37:20] @SI473x_setFrequency Frequency set to 7.074. KHz
[12:37:20] @tft_updateBand Band_slot(1)
[12:37:21] @SI473x_Setup Loading Si473x library (c) PU2CLR in SSB mode reset(1)
[12:37:21] @SI473x_Setup Si473X chip found at address 0x11
[12:37:21] @SI473x_Setup Device reset completed...
[12:37:22] @SI473x_Setup SSB patch loaded successfully (639 msec)
[12:37:22] @SI473x_Setup Antenna Capacitor set
[12:37:22] @SI473x_setFrequency Frequency set to 7.074. KHz
[12:37:22] @SI473x_Setup Frequency set slot(1)
[12:37:22] @SI473x_Setup set FrequencyStep as 1
[12:37:22] @SI473x_Setup Bandwidth index(2)
[12:37:22] @SI473x_Setup Cutoff filter set
[12:37:22] @SI473x_Setup set BFO as 975
[12:37:22] @SI473x_Setup set AGC as 0==0
[12:37:22] @SI473x_Setup Set automatic volume control
[12:37:22] @SI473x_Setup Set volume
[12:37:22] @SI473x_Setup current frequency read from Si473x is 7.074
[12:37:22] @SI473x_Status |AGC:AGC ON|LNA GAIN index:0/0|BW:3.0 KHz|S=6|SNR 0|RSSI 11 dBuV|Volume: 255|BFO +975|BFOStep: 25|freq=7.074|Step: 1|
[12:37:22] @setup Si4735 receiver initialized Band_slot(1)

EEPROM list

EEPROM list
|00010 -- a0 86 01 00 ff ff ff ff ff ff |..........|
|00020 -- ff ff ff ff ff ff ff ff ff ff |..........|
|00030 -- 64 00 00 00 ff ff ff ff ff ff |d.........|
|00040 -- 04 00 00 00 ff ff ff ff ff ff |..........|
|00050 -- 01 00 00 00 ff ff ff ff ff ff |..........|
|00060 -- 31 30 32 00 ff 66 00 00 00 ff |102..f....|
|00070 -- 43 54 37 41 42 41 00 00 00 00 |CT7ABA....|
|00080 -- 49 4e 36 30 00 00 00 00 ff ff |IN60......|
|00090 -- 47 61 69 61 7a 31 00 00 00 00 |Gaiaz1....|
|00100 -- 00 00 00 00 00 00 00 00 00 00 |..........|
|00110 -- 00 00 00 00 00 00 00 00 00 00 |..........|
|00120 -- 00 00 00 00 00 00 00 00 00 00 |..........|
|00130 -- 46 46 34 35 36 34 35 36 46 46 |FF456456FF|
|00140 -- 00 00 00 00 00 00 ff ff ff ff |..........|
|00150 -- 06 ff ff ff ff ff ff ff ff ff |..........|
|00160 -- 05 ff ff ff ff ff ff ff ff ff |..........|
|00170 -- 00 ff ff ff ff ff ff ff ff ff |..........|
|00180 -- 01 ff ff ff ff ff ff ff ff ff |..........|
|00190 -- 50 00 00 00 ff ff ff ff ff ff |P.........|
|00200 -- 3d 31 34 2e 30 37 34 7c 53 74 |=14.074|St|
|00210 -- 65 70 3a 20 31 7c ff ff ff ff |ep: 1|....|
|00220 -- 40 1f 00 00 ff ff ff ff ff ff |@.........|
|00230 -- 28 23 00 00 ff ff ff ff ff ff |(#........|
|00240 -- 00 00 00 00 00 00 00 00 ff ff |..........|
|00250 -- 2f 72 64 78 2e 74 78 74 00 00 |/rdx.txt..|
|00260 -- 00 00 00 00 00 00 ff ff ff ff |..........|
|00270 -- ff ff ff ff ff ff ff ff ff ff |..........|
|00280 -- ff ff ff ff ff ff ff ff ff ff |..........|
|00290 -- ff ff ff ff ff ff ff ff ff ff |..........|
|00300 -- ff ff ff ff ff ff ff ff ff ff |..........|
|00310 -- 37 33 20 61 6e 64 20 47 4c 00 |73 and GL.|
|00320 -- 00 00 00 00 00 00 ff ff ff ff |..........|
|00330 -- ff ff ff ff ff ff ff ff ff ff |..........|

[12:37:22] @setup -----------------------------------
[12:37:22] @setup Z=12:37 TZ=00:00 Local=12:37
[12:37:22] @setup *** Transceiver ready ***
[12:37:30] @ft8_run -------- RX w[0/60440]----------
[12:37:43] @ft8_run [00] w[0] 1812  -24 DK1XAM IW1ALW R-04
[12:37:45] @ft8_run -------- RX w[1/60352]----------
[12:37:46] @tft_checktouch Detected TFT touch x=110 y=17
[12:37:46] @tft_checktouch Detected TFT touch x=110 y=14
[12:37:58] @ft8_run [00] w[1] 2393  -24 CQ EG4SDC IN80
[12:37:58] @data_setup USB file (/rdx.txt) export started
[12:37:58] @tft_DataLoggerUSB USB export started
[12:38:00] @ft8_run -------- RX w[0/21104]----------
[12:38:13] @ft8_run [00] w[0] 1812  -24 DK1XAM IW1ALW 73
[12:38:15] @ft8_run -------- RX w[1/21104]----------

73 de Frederico